mstrens/grbl_controller_esp32 #5: Tips go wrong. Expect your help.

anshilike, 2019-10-17:
Hello, I'm a novice in ARDUINO. I am burning this program until esp32 is unsuccessful, showing whether "'Serial2' was not declared in this scope" is the problem of "change Serialx.flush() and number of Rx bytes before interrupt occurs"? If not, where would it be? I look forward to your help. Thank you！！！！

mstrens, 2019-10-18:
Normally the program is OK.
It is used by quite many people for the RS-CNC32
Here a link:
https://www.makerfr.com/en/

here a few points of attention:
- do you use a quite recent version of Arduino IDE
- do you install the library for the ESP32
- do you select a ESP32 board in arduino tool menu
You can find on internet how to do those points

anshilike, 2019-10-18:
> Normally the program is OK.
> It is used by quite many people for the RS-CNC32
> Here a link:
> https://www.makerfr.com/en/
>
> here a few points of attention:
>
> * do you use a quite recent version of Arduino [...]

I downloaded the latest version of Arduino and updated the official esp32 data. The program has been successfully burned!! It's very exciting. Tomorrow I will connect TFT and other PCB tests, thank you for your reply！

closed 2019-10-30 by mstrens