mstrens/grbl_controller_esp32 #7: Problem with TFT operation

maximius, 2019-11-15:
Good afternoon

I'm trying to assemble the device you designed. ESP32 successfully firmware through Arduino IDE. 

If after loading ESP 32 to include without connected TFT, the serial monitor works according to the program.
[image]

If I connect a TFT, a cyclic error appears in the serial monitor.
[image]

 Tried different versions of boards and memory volumes. Tried other boards as well. The result is the same. TFT 3.2" 320x240 ILI9341.
Connection according to the following scheme
[image]<br /><a target='_blank' href='https://ru.imgbb.com/'>image share</a><br />

mstrens, 2019-11-15:
The program works for sure. There are several users.

Continuous reset with ESP32 can occur when power supply is not strong/stable enough.
Wifi consumes a lot but only by peak.
Perhaps you are at the limit.
You can try 
- disabling the wifi (one parameter in config file)
- changing the power supply
- adding condensators on the 5V

maximius, 2019-11-18:
Found the cause of the error. It turned out that one GND pin next to the V5 pin did not connect to the rest of the GND. He squandered the tension. Thanks for the help.

closed 2019-11-18 by maximius