mstrens/grbl_controller_esp32 #8: When will the 4-axis version appear?

anshilike, 2019-11-18:
look forward to

mstrens, 2019-11-18:
The version is available on github but currently it is in another branch (AA-Axis).
This version can be configured for 3 or 4 axis depending on your set up in the config.h file.

I did not make many tests. So it could be that there are some bugs.
If so, let me know and I will try to fix them.

Note: If you are using a STM32, you have to activate the option AA-AXIS too.
Furthermore, if you already had uploaded a 3 axis version on the STM32, you have to reset your GRBL parameters (because the old parameters are kept but 4 axis uses more parameters and there is a shift)

anshilike, 2019-11-25:
I haven't found any problems with the simple test of esp32.

closed 2019-11-25 by mstrens