mstrens/grbl_controller_esp32 #12: Suggestion to use M5Stack ESP32 with built in 3 buttons and touch screen etc

splitn2, 2020-02-19:
Hi, have you seen the M5STACK esp32 units, very nicely packaged ready to program. I have one running esp32 port of GRBL but I would like to put a simple stack on top with the gcode sender built in for headless operation. 

I'll have a look at your code and see if I can work out how to help.

Have a look at www.m5stack.com

cheers
Rich

mstrens, 2020-02-19:
For you information, you can also buy for a good price a set of board that allows to use the ESP32 controller with a grbl version running on a STM32 and supporting 4 axes, laser, ...
You can also use just the ESP32 controller board and connect it to another board running GRBL (e.g. based on Avr328) .
Here the link
https://www.makerfr.com/en/cnc/grbl-32-bits-board/presentation-grbl-32bits-board/
On this site, you can purchase it, and find a lot of tutorial and info.

splitn2, 2020-02-20:
thanks @mstrens I'll have a look. I guess I like the M5STACK as its allready nicely integrated into a tidy wee case. I'll check the makerfr out! cheers, Rich

closed 2020-05-02 by mstrens