mstrens/grbl_controller_esp32 #14: Z-axis stops due to Alarm-1

jaygcnc, 2020-03-18:
So I have a few issues going on. Any insight on how to resolve them would be greatly appreciated. 

My alarm code is "Alarm-1" reset with homing
Also on startup sometimes I get a Door:0 error , but it goes away when I reset the power.

This is my main issue. While using the nun chuck or the touchscreen the x-axis and y-axis move correctly, but the Z-axis will move only a little bit then trigger the alarm.  It was working properly, but now I can barley get it to move without triggering the alarm.

This is another issue I am having. When I hit home all of the axis will move in the correct order and stop when triggering end stops. However, sometimes the x-axis will stop before any of the end stops are reached. Once the y-axis is reached the x-axis will move toward the end stop. Most of the time it will trigger the ends stop, but sometimes it will give up.

mstrens, 2020-03-18:
The issues you report are not related to the controller but to GRBL itself.
It looks like you have noise on your endstop signal and other pin (like Door).
Do you use shielded cables for you end stops?
If not I recommend using such cable. You can also use an optocoupler module for you end stops.
There is a parameter in GRBL that you can use to increase noise protection for the endstops (probably in config.h file)

jaygcnc, 2020-03-19:
Yes, I used shielded cable for the end stops. My end stops are wired normally closed. I didn't see a parameter in the config.h file to increase noise protection.  Then again I am very new at coding.

I think I am going to go the optocoupler module route. There are two different basic types that I see. One looks like it can just get wired into the end stops. The JBtek one looks more advanced. I was wondering if I need to add a code in the GRBL to accept the optocoupler module? Thank you for your help. I appreciate you taking the time to help me out. 

[https://www.amazon.com/Cylewet-Optocoupler-Mounting-Coupler-Arduino/dp/B06X6M2NNY/ref=sr_1_16_sspa?keywords=optocoupler+module&qid=1584657277&s=electronics&sr=1-16-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyWTREQ0QwS0dTNlVEJmVuY3J5cHRlZElkPUEwNTUwNzA2MTNXWlNQRERTRE9FQyZlbmNyeXB0ZWRBZElkPUEwMTY0ODczMUxLQjhEWEEwUlg1WCZ3aWRnZXROYW1lPXNwX210ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=](url)

[https://www.amazon.com/JBtek-Channel-Module-Arduino-Raspberry/dp/B00KTEN3TM/ref=sr_1_9?keywords=optocoupler+elegoo&qid=1584657802&s=electronics&sr=1-9&swrs=130392AC71F7C229A904A8AC49296427](url)

mstrens, 2020-03-20:
Do not use a module with relay. It will delay the signal.
I think you can find better module like this:
https://fr.aliexpress.com/item/32944876416.html?spm=a2g0o.productlist.0.0.272b35b4ZqxJ4c&algo_pvid=b0f0f804-d820-4e2c-9831-1dda983349e7&algo_expid=b0f0f804-d820-4e2c-9831-1dda983349e7-1&btsid=0ab50f6115846890971965416eb578&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
or
https://fr.aliexpress.com/item/32964487766.html?spm=a2g0o.productlist.0.0.272b35b4ZqxJ4c&algo_pvid=b0f0f804-d820-4e2c-9831-1dda983349e7&algo_expid=b0f0f804-d820-4e2c-9831-1dda983349e7-3&btsid=0ab50f6115846890971965416eb578&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
I expect it should work; still I have no experience with them

jaygcnc, 2020-03-20:
Thank you for the advice. I will try one out and let you know how if it works.

closed 2021-05-20 by mstrens