mstrens/grbl_controller_esp32 #16: Parser issue, not catching wposOrMpos correctly, breaks handleLastNumericField()

petervanderwalt, 2020-04-25:
Referring to https://github.com/mstrens/grbl_controller_esp32/blob/bca2837d8c2ed7b5ebb0edefc5347c77f52dc9a7/com.cpp#L195-L201

Traced the issue though to the linked code above

If I just add Serial.print to debug, I can see why it fails:  Just havent dug in deep enough into your code to fashion a fix (will if you don't get time, but I bet you'll have an idea off-hand before I even get to digging deeper (: 

Added my serial.println's:
```
case ':' :
      if ( getGrblPosState == GET_GRBL_STATUS_WPOS_HEADER ) { // separateur entre field name et value 
         getGrblPosState = GET_GRBL_STATUS_WPOS_DATA ; 
         wposOrMpos = strGrblBuf[0] ;       // save the first char of the string that should be MPos or WPos
         Serial.println("");
         Serial.println(strGrblBuf);
         Serial.println("");

         strGrblIdx = 0 ;
         
         strGrblBuf[strGrblIdx] = 0 ;
         wposIdx = 0 ;
      }

```
And I can see 

```
<Idle|WPos:

Idle

0.000,0.000,0.000|FS:0,0>

```
That at that point where you store wposOrMpos = strGrblBuf[0] ; that strGrblBuf still contains Idle/Jog/Hold/etc string, and not the expected wPos/mPos 

Subsequently, handleLastNumericField() breaks the calculation as it checks:

```
 if ( wposOrMpos == 'W') {                  // we got a WPos
            wposXYZA[wposIdx] = temp ;
          } else {                                   // we got a MPos
            wposXYZA[wposIdx] = temp - wcoXYZA[wposIdx] ;
          }
```
But wposOrMpos = "i" so we always hit the "else" calculation (: 

Picked up that wpos/mpos data isnt coming though correctly if I use coordinate offsets, along with $10=0 (wpos reporting) 
With the default $10=1 it "works" mPos reporting gets handled by the "else" section of handleLastNumericField.   But if reporting is not set to $10=1, then we still hit "else" even though it shouldnt.  The section of code in handleLastNumericField clearly intends to handle both reporting settings, its just that earlier it gets stored the wrong character (:

petervanderwalt, 2020-04-25:
Got it!

Right, so 

https://github.com/mstrens/grbl_controller_esp32/blob/bca2837d8c2ed7b5ebb0edefc5347c77f52dc9a7/com.cpp#L269-L273

Was should also include ``` || getGrblPosState == GET_GRBL_STATUS_WPOS_HEADER ```

ie change to 

```
} else if ( ( (c >= 'A' && c <= 'Z') || (c >= 'a' && c <= 'z') ) && ( getGrblPosState == GET_GRBL_STATUS_START || getGrblPosState == GET_GRBL_STATUS_HEADER ||
                    getGrblPosState == GET_GRBL_STATUS_CLOSED || getGrblPosState == GET_GRBL_STATUS_MESSAGE || getGrblPosState == GET_GRBL_STATUS_WPOS_HEADER ) ) { 
             strGrblBuf[strGrblIdx++] = c ;
             strGrblBuf[strGrblIdx] = 0 ;
          }
```

Now it append Mpos/Wpos to strGrblBuf correctly

Next at https://github.com/mstrens/grbl_controller_esp32/blob/bca2837d8c2ed7b5ebb0edefc5347c77f52dc9a7/com.cpp#L176-L178 after copying the string to machineStatus, I cleaned up strGrblBuf and strGrblIdx :

```
if ( getGrblPosState == GET_GRBL_STATUS_START ) {
           getGrblPosState = GET_GRBL_STATUS_WPOS_HEADER ; 
           memccpy( machineStatus , strGrblBuf , '\0', 13);        
           strGrblIdx = 0;
           strGrblBuf[strGrblIdx] = 0 ;
```

And now wposOrMpos = strGrblBuf[0] and strGrblBuf contains Mpos or WPos correctly as expected

petervanderwalt, 2020-04-25:
Tested good too! 

I get a WPos feedback, with a WCO entry.  And calculated MPOS correctly 

<Idle|WPos:0.000,0.000,0.000|FS:0,0|WCO:92.188,0.000,0.000>
WCO -  X:92.19 Y:0.000.00 A:0.00
WPOS -  X:0.00 Y:0.00 Z:0.00 A:0.00
MPOS -  X:92.19 Y:0.00 Z:0.00 A:0.00

mstrens, 2020-04-25:
Thanks for finding and fixing the issue.
I will include the correction in the next release (I am currently preparing a solution using sd files in order to let the user force a new touch screen calibration and changing Wifi parameter without having to reflash)

stratogk, 2020-04-26:
> Thanks for finding and fixing the issue.
> I will include the correction in the next release (I am currently preparing a solution using sd files in order to let the user force a new touch screen calibration and changing Wifi parameter without having to reflash)

Hello i will sugest for the wifi settings to set by yourself the ip address (Dhcp or fix ip)
thank you

mstrens, 2021-05-20:
Fix ip is supported in newer versions.

closed 2021-05-20 by mstrens