mstrens/grbl_controller_esp32 #20: SDcard write problem

aik3n, 2020-07-28:
Hello, first please your work and your effort.
I have problems when uploading a file, a "sdcard protect?" Message appears on the web interface, I have looked at the code but it is difficult for me, I have seen that if there is a file with the same name it first deletes them, and that it does ok, but there is some problem with writing the file, the sd card is sd hc 4gb fat32 format, I have checked the wiring and it seems correct, when I download a file from the sd it works correctly.
Any suggestion?

Thanks in advance

mstrens, 2020-07-28:
The issue is related to the version of SDFat being used.

In the file  browser.cpp you have currently the folowing lines.

#define MS_WRITE 0b00010101 // we have to define this to open file because there is a conflict between ESP32 and SdFat lib about the code to be used in open cmd
                           // this is the value used by SdFat in version 1.1.0 for read, Write, create, at_end

//#define MS_WRITE 0b1100011 // we have to define this to open file because there is a conflict between ESP32 and SdFat lib about the code to be used in open cmd
                           // this is the value used by SdFat in version 1.0.7

If you use a version of SdFat 1.0.7 or higher, you shoud put the first #18 define as comment and activate the second #define
So, you should have:
//#define MS_WRITE 0b00010101 // we have to define this to open file because there is a conflict between ESP32 and SdFat lib about the code to be used in open cmd
                           // this is the value used by SdFat in version 1.1.0 for read, Write, create, at_end

#define MS_WRITE 0b1100011 // we have to define this to open file because there is a conflict between ESP32 and SdFat lib about the code to be used in open cmd
                           // this is the value used by SdFat in version 1.0.7

aik3n, 2020-07-30:
Hello again,

Thanks for the information, I have been able to correct the problem, I had version 1.1.4 installed and no option works well, I have installed 1.1.0 and everything works fine.

Best regards,

kaktus312, 2020-12-07:
Hi! 
I have version 1.1.4 and the same error too. 
I changed the value on line #18 in the file browser.cpp on value 0b01010101 (now it looks like #define MS_WRITE 0b01010101) and it seems everything is ok!

closed 2021-05-20 by mstrens