mstrens/grbl_controller_esp32 #25: No movement in Auto mode

kaktus312, 2020-12-08:
"Move" screen...
The button "Auto" is displayed, but if i press any button (X,Y,Z) there are no movements. In other modes (0.01/0.1/1/10) cnc moves correctly 
What will i have to change in my script (maybe delays somewere) to fix it.

mstrens, 2020-12-08:
If you do not change the code, it should works.
Still you have press for some time. The longer you press the longer the move becomes.
It is the same process as for using the nunchuk

kaktus312, 2020-12-09:
It seems the problem solved. I have changed line #581 of the file com.cpp 
from
 if (bufferAvailable[0] > 15) {    // tests shows that GRBL gives errors when we fill to much the block buffer
to
 if (bufferAvailable[0] >= 5) {    // tests shows that GRBL gives errors when we fill to much the block buffer

closed 2020-12-09 by kaktus312

mstrens, 2020-12-09:
Which version of GRBL where you running?
Is it the version foreseen for a stm32f103 (which have extended buffers) used in RS-CNC32 project or the original version foreseen for an arduino (avr328)?

kaktus312, 2020-12-09:
Hi! I use MKS DLC with Atmega328P on board and grbl 1.1. 
It has 15 block buffer, that's why there were no movements with code like ...bufferAvailable[0]>15...
I tryed to change it to ...bufferAvailable[0]>=15..., but the movements were intermittent and not stabile.

mstrens, 2020-12-09:
I normally use the program with a stm32f103 where I extended the buffer because it has more ram.
That is probably the reason why it worked.
With a avr328, it seems that the test on ESP32 has to be adapted in the way you did it.
Thanks for the feedback