mstrens/grbl_controller_esp32 #27: 'File' does not name a type - Error when compiling

PhilippEBZ, 2020-12-14:
Hi,

I tried to compile the project with Arduino 1.8.13. At first I didnt have the SdFat library which I installed, but now I get this error and i dont know how to make it work. Do you have some ideas?

Here is the output I get:

Thank you!!

menu_file.cpp:53:1: error: 'File' does not name a type
 File fileToShow ;   // this contains the file being sent from SD
 ^
C:\TEMP\arduino_build_855381\sketch\menu_file.cpp: In function 'uint16_t fileCnt(uint8_t)':
menu_file.cpp:138:3: error: 'File' was not declared in this scope
   File file ; 
   ^
C:\TEMP\arduino_build_855381\sketch\menu_file.cpp:138:3: note: suggested alternative:
In file included from C:\TEMP\arduino_build_855381\sketch\TFT_eSPI_ms/TFT_eSPI.h:114:0,
                 from C:\TEMP\arduino_build_855381\sketch\config.h:4,
                 from C:\TEMP\arduino_build_855381\sketch\menu_file.cpp:13:
C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src/FS.h:47:7: note:   'fs::File'
 class File : public Stream
       ^
menu_file.cpp:140:11: error: 'file' was not declared in this scope
   while ( file.openNext( &aDir[level] ) ) {
           ^
menu_file.cpp:145:3: error: 'file' was not declared in this scope
   file.close();
   ^
C:\TEMP\arduino_build_855381\sketch\menu_file.cpp: In function 'boolean updateFilesBtn()':
menu_file.cpp:165:3: error: 'File' was not declared in this scope
   File file ;
   ^
C:\TEMP\arduino_build_855381\sketch\menu_file.cpp:165:3: note: suggested alternative:
In file included from C:\TEMP\arduino_build_855381\sketch\TFT_eSPI_ms/TFT_eSPI.h:114:0,
                 from C:\TEMP\arduino_build_855381\sketch\config.h:4,
                 from C:\TEMP\arduino_build_855381\sketch\menu_file.cpp:13:
C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src/FS.h:47:7: note:   'fs::File'
 class File : public Stream
       ^
menu_file.cpp:168:12: error: 'file' was not declared in this scope
     if ( ! file.openNext( &aDir[dirLevel] ) ) {       // ouvre le prochain fichier dans le répertoire courant ; en cas d'erreur, retour à la page info avec un message d'erreur 
            ^
menu_file.cpp:174:5: error: 'file' was not declared in this scope
     file.close() ;
     ^
menu_file.cpp:182:12: error: 'file' was not declared in this scope
     if ( ! file.openNext( &aDir[dirLevel] ) ) {
            ^
menu_file.cpp:189:10: error: 'file' was not declared in this scope
     if ( file.isDir() ) { 
          ^
menu_file.cpp:193:12: error: 'file' was not declared in this scope
     if ( ! file.getName( pfileNames , 21 ) ) {   // Rempli fileNames avec le nom du fichier
            ^
menu_file.cpp:203:5: error: 'file' was not declared in this scope
     file.close() ;
     ^
C:\TEMP\arduino_build_855381\sketch\menu_file.cpp: In function 'boolean setFileToRead(uint8_t)':
menu_file.cpp:257:3: error: 'fileToShow' was not declared in this scope
   fileToShow.close() ;
   ^
C:\TEMP\arduino_build_855381\sketch\menu_file.cpp: In function 'void setShowBuffer()':
menu_file.cpp:376:3: error: 'fileToShow' was not declared in this scope
   fileToShow.seek( sdMillPos ) ; // set the same position for the file used for display
   ^
C:\TEMP\arduino_build_855381\sketch\menu_file.cpp: In function 'void setPrevShowBuffer()':
menu_file.cpp:421:3: error: 'fileToShow' was not declared in this scope
   fileToShow.seek( sdShowBeginPos ) ; // set this position for the file used for display
   ^
C:\TEMP\arduino_build_855381\sketch\menu_file.cpp: In function 'void setNextShowBuffer()':
menu_file.cpp:432:3: error: 'fileToShow' was not declared in this scope
   fileToShow.seek( sdShowBeginPos ) ; // set this position for the file used for display
   ^
browser.cpp:38:1: error: 'File' does not name a type
 File root ; // used for Directory 
 ^
C:\TEMP\arduino_build_855381\sketch\browser.cpp: In function 'void DownloadFile(String)':
browser.cpp:327:7: error: 'File' was not declared in this scope
       File download ;
       ^
C:\TEMP\arduino_build_855381\sketch\browser.cpp:327:7: note: suggested alternative:
In file included from C:\TEMP\arduino_build_855381\sketch\TFT_eSPI_ms/TFT_eSPI.h:114:0,
                 from C:\TEMP\arduino_build_855381\sketch\config.h:4,
                 from C:\TEMP\arduino_build_855381\sketch\browser.cpp:12:
C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src/FS.h:47:7: note:   'fs::File'
 class File : public Stream
       ^
browser.cpp:328:7: error: 'download' was not declared in this scope
       download = sd.open( filename.c_str() );
       ^
C:\TEMP\arduino_build_855381\sketch\browser.cpp: At global scope:
browser.cpp:356:1: error: 'File' does not name a type
 File UploadFile;
 ^
C:\TEMP\arduino_build_855381\sketch\browser.cpp: In function 'void handleFileUpload()':
browser.cpp:366:9: error: 'UploadFile' was not declared in this scope
         UploadFile.close() ;
         ^
browser.cpp:372:12: error: 'UploadFile' was not declared in this scope
         if(UploadFile) { 
            ^
browser.cpp:381:12: error: 'UploadFile' was not declared in this scope
         if(UploadFile && ( errorWhileUploading == false) )          // If the file was successfully created
            ^
C:\TEMP\arduino_build_855381\sketch\browser.cpp: In function 'void sd_dir()':
browser.cpp:403:5: error: 'root' was not declared in this scope
     root.close() ;
     ^
C:\TEMP\arduino_build_855381\sketch\browser.cpp: In function 'void printDirectory(const char*, uint8_t)':
browser.cpp:428:3: error: 'File' was not declared in this scope
   File root1 = sd.open(dirname);
   ^
C:\TEMP\arduino_build_855381\sketch\browser.cpp:428:3: note: suggested alternative:
In file included from C:\TEMP\arduino_build_855381\sketch\TFT_eSPI_ms/TFT_eSPI.h:114:0,
                 from C:\TEMP\arduino_build_855381\sketch\config.h:4,
                 from C:\TEMP\arduino_build_855381\sketch\browser.cpp:12:
C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src/FS.h:47:7: note:   'fs::File'
 class File : public Stream
       ^
browser.cpp:431:7: error: 'root1' was not declared in this scope
   if(!root1){
       ^
browser.cpp:435:7: error: 'root1' was not declared in this scope
   if(!root1.isDir()){
       ^
browser.cpp:439:3: error: 'root1' was not declared in this scope
   root1.rewind();
   ^
browser.cpp:441:8: error: expected ';' before 'file1'
   File file1 ;
        ^
browser.cpp:442:9: error: 'file1' was not declared in this scope
   while(file1.openNext(&root1)){
         ^
browser.cpp:468:3: error: 'file1' was not declared in this scope
   file1.close();
   ^
C:\TEMP\arduino_build_855381\sketch\browser.cpp: In function 'void SD_file_delete(String)':
browser.cpp:482:5: error: 'File' was not declared in this scope
     File dataFile = sd.open( filename.c_str() ); //  
     ^
C:\TEMP\arduino_build_855381\sketch\browser.cpp:482:5: note: suggested alternative:
In file included from C:\TEMP\arduino_build_855381\sketch\TFT_eSPI_ms/TFT_eSPI.h:114:0,
                 from C:\TEMP\arduino_build_855381\sketch\config.h:4,
                 from C:\TEMP\arduino_build_855381\sketch\browser.cpp:12:
C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS\src/FS.h:47:7: note:   'fs::File'
 class File : public Stream
       ^
browser.cpp:483:9: error: 'dataFile' was not declared in this scope
     if (dataFile) {
         ^
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI  wird verwendet
Bibliothek FS in Version 1.0 im Ordner: C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS  wird verwendet
Bibliothek SPIFFS in Version 1.0 im Ordner: C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS  wird verwendet
Bibliothek Wire in Version 1.0.1 im Ordner: C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Wire  wird verwendet
Bibliothek SdFat in Version 2.0.2 im Ordner: C:\Users\pkleesse\Documents\Arduino\libraries\SdFat  wird verwendet
Bibliothek Preferences in Version 1.0 im Ordner: C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Preferences  wird verwendet
Bibliothek WiFi in Version 1.0 im Ordner: C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi  wird verwendet
Bibliothek WebServer in Version 1.0 im Ordner: C:\Users\pkleesse\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer  wird verwendet
exit status 1
'File' does not name a type

mstrens, 2020-12-14:
I expect the sdfat library is not correctly installed.
Take also care of the version that you install (otherwise you have to make a minor change in a file).
Look in the tutorial of RS-CNC 32 on this side to have more detail
 https://www.makerfr.com/forum/viewtopic.php?f=38&t=6446&start=240

PhilippEBZ, 2020-12-14:
Hi! Thank you for the fast reply!

You were right, I didn`t have that document and I didnt know that the newest SdFat Lib from the Arduino Library Manager wouldn´t work. 

I could succefully compile and upload through the Arduino IDE.

I hooked up an SD Card reader to the pins in the decription on Github and set the CS pin accordingly. I also set the wifi settings to AP. 
Because I dont have the TFT and neither the nunchuk and just want to try the browser interface, sd card and serial pass through I commented the following:

- tftInit()
- initButtons()
- nunchuk_init()

out of the setup() routine. 

Unfortunatly, the ESP32 is rebooting over and over. 

Is it possible to test the grbl_controller without tft and nunchuk or would that be impossible?

Is there a away over a define maybe to turn off those features which I (and maybe others) are not using?

In general I really like the idea of your project!!! Thank you for the help

This is what I get out of the terminal:

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

using config.h param for WiFi
Wifi config will be executed
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060630  A0      : 0x800d7921  A1      : 0x3ffb1f70  
A2      : 0x3ffc1edd  A3      : 0x3ffc33c4  A4      : 0x3ffc3a40  A5      : 0x00000000  
A6      : 0x400d2b4c  A7      : 0x0101a8c0  A8      : 0x800d50fa  A9      : 0x3ffc180c  
A10     : 0x00000000  A11     : 0x3f402019  A12     : 0x3ffc33c4  A13     : 0x3ffc33a1  
A14     : 0x3ffb1f4c  A15     : 0x3ffc3c28  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400e20b4  LEND    : 0x400e2116  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb1f70 0x400d791e:0x3ffb1f90 0x400e4205:0x3ffb1fb0 0x4008939d:0x3ffb1fd0

Rebooting...
```

mstrens, 2020-12-14:
It is possible to use the ESP32 without the TFT.
If you just want to test the wifi (and access to the SD card via the wifi), you do not have to modify the original firmware.
If you want to test the serial pass through you have to change just one line of code.

To do so, change in the file grbl_controller_esp32.ino, in line 124 there is currently:
uint8_t statusPrinting = PRINTING_STOPPED ;

You can change it to:
uint8_t statusPrinting = PRINTING_FROM_USB ;

When this version is flashed in the ESP32, it will work like you manually entered in the print using USB mode.
So, il you connect the ESP32 board to GRBL, you should be able to communicate directly from pc to GRBL using e.g. the Arduino serial terminal (e.g. to send a command like $$ to get the GRBL parameter). Do not forget to set the baudrate in arduino serial terminal on 115200 baudrate

PhilippEBZ, 2020-12-15:
Hi,

I tried that, changed statusPrinting to PRINTING_FROM_USB and uncommeted those things I mentioned above, but than I just get the serial output:
`failed to open calibrate.txt
`
There was opening and closing a wifi AP with the name I defined before when I had those things in a comment and it was rebooting all the time. I couldnt access the AP because it was gone every 2s but I could see it in my phone. Now there is no AP showing up.

So I put a comment on tft_init() again and now it succesfully initialized and opend the AP! :-)
It also connected to my Arduino Nano running GRBL 1.1h.

Unfortunatly, after connecting with the AP, i couldnt access a webpage under the IP i gave in config.h.
I even set my Wifi Card in the laptop to the same address range with a fixed IP, but it still wouldnt connect.  

Do you have any more ideas?

Greetings
Philipp

mstrens, 2020-12-15:
It is strange.
Other users said me that just changing one line of code in the original firmware was enough and no body reported an issue with calibrate.txt. I think that this text is just informative (I probably forgot to put it as comment after debugging).

I sea 2 reasons for the ESP32 rebooting:
- no enough power.  This is mainly when wifi is used. It can consume up to 1A for a short time. If you power it via USB, it could be that your PC provides only 0.5A (as specified in USB spec). Then use a separate power supply.
- a  conflict in the use of some ressources due to a wrong type of board when compiling