mstrens/grbl_controller_esp32 #28: TFT problem

krucho, 2021-01-14:
Hi! I'am using a 2.8 inches 320x240 display, same hardware and wiring of the code here.
I can't see anything on the screen, for testing the hardware I used the original TFT_eSPI library and works as expected. When I tried to replace the library I noticed that there are some differences and the code don't compile.
Any idea on what is happening? 
Thanks!

mstrens, 2021-01-14:
I made some changes to the library e.g. to add some special characters (for french) and also to (I hope so) improve the handling of the touch screen.
Therefore you do not have to use the original library but the code I provided on github.
Those files have to be installed on your pc using exactly the same directory structure as the one given here (on github); so tft files are in a subdirectory.
When the original library is installed, it could be that there is a conflict with the code I provide (but I do not think so).

If you need more support, you should provide the compilation errors.

krucho, 2021-01-14:
Thanks for your reply! There are no compilation errors when I use your code (and the same directory structure), the problem is the TFT shows nothing. I used the original library examples just for checking the hardware and wiring with success.

mstrens, 2021-01-14:
Normally my code works because I am on a cnc forum and it is used by many members.
1) Are you sure that your wiring is as specified in my readme section.
It could be that the pins being used are different from the standard lib default.
2) Some members where not able to run the code because when compiling they where using a wrong ESP32 board. They did not get compilation error but the code was not running (probably because there are some internal conflict between some function). When they tried compiling another board (e.g. wmos lolin32) they run the code successfully.
3) some members had issues because they tried to use a tft board that did not had the IC used to manage the touchscreen.

krucho5, 2021-01-14:
1) Yes, I'm sure the wiring is ok cause I modified the standard lib in order to use the same pinout as you and the display works.
2) I will try with another dev board and/or board selection, I'm using a SparkFun ESP32 thing now.
3) The TFT uses the ILI9341.

To check the wiring and the SPI working, I also used an external SD card and everything went just fine.
As I mentioned, will try with another ESP32. Thanks again.

mstrens, 2021-01-14:
I presume you understood me but just for clarity: I do not say that you have to try another dev board but only that you can try to select another board in the arduino IDE.

Please note that the ILI9341 is only to manage the TFT but not the touchscreen.
When the TFT has a touchscreen, it must have a small IC (about 10/16 pins) on the back side.
The same TFT module exists with and without this IC.

krucho5, 2021-01-15:
Just another question, which version of the ESP32 package are you using to compile? In the MakerFr tutorial I saw the 1.0.1 version, currently I'm using 1.0.3, I will change it for the next try.

mstrens, 2021-01-15:
Î use currently 1.0.4.

krucho, 2021-01-15:
Well, finally the set up is working!
Using an ESP32 Wrover module, I can't make it with the ESP32 Thing.
Thank you for your reply and congratulations for the great work! I will close the issue

closed 2021-01-15 by krucho