mstrens/grbl_controller_esp32 #33: When I repeatedly press Pause/Resume several times,it will stop sending gcode.

viewsat, 2021-05-08:
Hello there:
When I repeatedly press Pause/Resume several times, it will stop sending gcode. Is this an error in my operation? Or is there an error in grbl_controller_esp32?
[image]

viewsat, 2021-05-10:
Pause Button test video
[https://youtu.be/5V1ZVn9fpcc](url)

mstrens, 2021-05-10:
Thanks for reporting this.
I expect I know the reason.
I will try to find a solution.

mstrens, 2021-05-12:
I put a new version on github that should fix this issue (I hope)

viewsat, 2021-05-13:
> I put a new version on github that should fix this issue (I hope)

I tested the pause (hold) button of the new version, and there will be no problem of stopping sending gcode.
Thank you very much!

closed 2021-05-20 by mstrens