mstrens/grbl_controller_esp32 #34: bCNC Telnet connection

thinkJD, 2021-05-14:
Hello,
thanks for your great work! I really like the project. 

I designed my own PCB, uploaded the newest firmware (12.5.) and installed it on my Woodpecker CNC board. 

Manual Jog works, Nunchuck works, Direct USB seems to reset the ESP32 but also works after reboot and Print -> USB

But:
I can’t connect bCNC via telnet (saw your issues on the bCNC project. So I used socket:// like in the readme described). 
I can connect, the WiFi logo turns green. Then I switch to print -> telnet and try to jog manually (in bCNC) but nothing happens. 

This happens on the release version and also on the current master.

I use the current bCNC installed via Pip in my Mac.

Any ideas why?

mstrens, 2021-05-14:
You are not the first one having an issue using bcnc.
If I remember well, someone had similar issue but using the USB connection
Can you use bcnc via the USB connection?

It could be that the ESP32 that manages the wifi generates some extra messages that are not accepted by bcnc.
Still, in  the last version, I tried to remove all kind of messages or I put them in a sequence like grbl ( [MSG:.......]) hoping it should just be discarded by grbl)

On what kind of MCU are you running grbl (an arduino or on another mcu like esp32, STM32)?

thinkJD, 2021-05-15:
I use a [Woopecker CNC board](http://wiki.sainsmart.com/index.php/101-60-280). It's a 8 Bit Atmega, maybe some Arduino clone with integrated stepper drivers. There is also an extension Port with 5v, reset and rx/tx for a external controller. I used this connector for grbl_controller_esp32. 

grbl_controller_esp32 and bCNC over USB works fine, but maybe not as expected. As soon as i connect bCNC over USB grbl_controller_esp32 restarts. Then I have to select print -> USB and it works fine. I can stop USB and Jog manually and switch back, no issues. So every connect / disconnect triggers a restart of grbl_controller_esp32. 

Tho only thing not working is the telnet connection. Strange.

thinkJD, 2021-05-19:
I installed the newest Version (17.05.) and tried again. Sadly without luck. This is the terminal Output from bCNC.

I did the following:

1. select print -> telnet
2. open bCNC
3. connect bCNC via socket
4. jog in some directions

## bCNC -> grbl_controller_esp32
```
$#
$G
$J=G91 X10.0 F100000
$J=G91 X-10.0 F100000
$J=G91 Y10.0 F100000
$J=G91 Y-10.0 F100000
$J=G91 Z1.0 F100000
$X
b'$G\n'
```

## grbl_controller_esp32 -> bCNC
```
<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:50.626,0.000,-3.736>
<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
<Idle|MP
os:0.000,0.000,0.000|Bf:15,128|FS:0,0>
os:0.000,0.000,0.000|Bf:15,128|FS:0,0>
os:0.000,0.000,0.000|Bf:15,128|FS:0,0>
os:0.000
,0.000,0
.000|Bf:15,128|FS:0,0>
os:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:50.626,0.000,-3.736>
os:0.000,0.000,0.000|Bf:15,128|FS:0,0>
os:0.000
,0.000,0
.000|Bf:15,128|FS:0,0|WCO:50.626,0.000,-3.736>
```

thinkJD, 2021-05-20:
I can confirm. I am using

Arduino IDE: v1.8.14
ESP32: v1.0.4

mstrens, 2021-05-20:
Thanks for the info.
I will try to make a test when I have time.