mstrens/grbl_controller_esp32 #37: When I set A axis=0, it becomes Z axis=0.

viewsat, 2021-05-23:
When I set A axis=0, it becomes Z axis=0.
see youtube
https://www.youtube.com/watch?v=t4HGG-T2hoM

mstrens, 2021-05-23:
Thanks for reporting the bug.
There is a copy/past bug.
In file config.h, there is currently:
#define _SETA_STRING "G10 L20 P1 A0\n"

It should be
#define _SETA_STRING "G10 L20 P1 Z0\n"

I will fix it in next update but you can easily fix it immediately in your version

viewsat, 2021-05-23:
> Thanks for reporting the bug.
> There is a copy/past bug.
> In file config.h, there is currently:
> #define _SETA_STRING "G10 L20 P1 A0\n"
> 
> It should be
> #define _SETA_STRING "G10 L20 P1 Z0\n"
> 
> I will fix it in next update but you can easily fix it immediately in your version

I change config.h follow your instructions, no problem.

Before modification => #define _SETA_STRING "G10 L20 P1 Z0\n"
After modification => #define _SETA_STRING "G10 L20 P1 A0\n"