mstrens/grbl_controller_esp32 #38: When I press Set...axis, the message on the top left looks wrong.

viewsat, 2021-05-24:
When I press Set...axis, the message on the top left looks wrong.
See youtube => https://youtu.be/_ZMQTXIZa6c

[image]

mstrens, 2021-05-24:
I do not think it is a bug.
The firmware is supposed to display here the last message/warning being generated.
So it is not always related to the action you take.
The message is cleared automatically when you go back from Setup screen to Info screen.

viewsat, 2021-05-24:
> 我不認為這是一個錯誤。
> 固件應該在此處顯示最近生成的消息/警告。
> 因此，它並不總是與您採取的行動有關。
> 從設置屏幕返回到信息屏幕時，該消息將自動清除。

> I do not think it is a bug.
> The firmware is supposed to display here the last message/warning being generated.
> So it is not always related to the action you take.
> The message is cleared automatically when you go back from Setup screen to Info screen.

Thank you for your quick response and explanation.