mstrens/grbl_controller_esp32 #39: USB stops working on GRBL when ethernet cable is connected between Controller ESP32 and GRBL ESP32

cotepat, 2021-06-01:
I use CNCjs on the GRBL board, connected directly with the USB port. As soon as I connect the Controller board using the ethernet cable, USB connection starts to act weird (it receives some messages but don't send any commands). This is true **even when connected with bluetooth or telnet** !

mstrens, 2021-06-01:
GRBL ESP32 use the same UART to communicate with USB and with the Serial TFT.
So there can conflict between USB and Serial TFT.
In previous version, TFT kept Serial opened even when BT or Telnet where used.
I just made a change in order to completely desactivate Serial TFT when BT or Telnet is activated.
I did not tested the solution. So please let me know if it is now ok or not in new version vers.2.1.d

cotepat, 2021-06-01:
Unfortunately, it did not seem to change anything... Also I saw the change in the code, but version was still at 2.1.c (not d)

mstrens, 2021-06-02:
Can you make the following test (with the version from yesterday).
Select bluetooth or telnet communication on TFT. Power the TFT Off and On again.
Check that the communication on TFT is bluetooth or Telnet.
I expect that when the TFT starts directly in BT or Telnet, the TFT ESP32 will not drive the Tx pin and so should avoid a conflict with USB on GRBL side.

In the version from yesterday, I added a Serial End function when bluetooth or telnet is activated but perhaps that this standard End function does not release the Serial Tx Pin when it has already been activated .

cotepat, 2021-06-02:
Yeah I did try that. And I also re-tried just now... Still the same. I confirm the connection started directly on telnet. Is it possible that the original state of grbllink is still GRBL_LINK_SERIAL due to that line in _com.cpp_ ?

`uint8_t grblLink = GRBL_LINK_SERIAL ;`

mstrens, 2021-06-02:
GRBL_LINK_SERIAL is the default value.
Still, normally the firmware retrieves the stored values (in preference) before it initialises the Serial port.
So if the preference is Telnet, the firmware should (with the version of yesterday) let Serial port in high impedance.

Still the firmware call a function uartDetach provided by Espressif. I should check what happens in this function.

cotepat, 2021-06-02:
For your information. As soon as I disconnect the TFT, I regain normal access to usb of grbl... I wonder if calling Serial end from a button for instance would also work.....

mstrens, 2021-06-02:
I made some more changes. I put the version on github.
Can you try it.
Now, I force the Tx pin in Input mode after the Serial End.
I also removed some code about Serial2 in the Setup (I forgot it yesterday).
I hope it will work.

cotepat, 2021-06-02:
Ok... Things have changed now. It is still not working but at least there is an important finding : it is possible to have usb working while ethernet cable is connected. I'll explain:

- When powering on with telnet or bluetooth : USB works **but telnet or bluetooth does not connect (status is ??)**
- When switching to serial, TFT works (on serial) but USB stops (this is expected)
- When switching back to telnet or bluetooth, TFT status becomes "??" and USB does not work.

mstrens, 2021-06-02:
It is strange. I do not understand what happens.
I will have to test it with some debug code.

cotepat, 2021-06-03:
Ok. I experimented a little more... Some really strange behaviour! Like sometimes bluetooth or telnet do connect and some things work (like the home button for instance) but the "move" arrows completely crash the TFT (everything hangs and I need to reset TFT).

mstrens, 2021-06-03:
Do you have a good telnet connection?
Could it be that the communication is quite bad and that some data are lost.
When TFT send one line of data to GRBL, it waits for an "OK" message before sending the next line.
A Move from TFT send more lines than just a Home.
Normally when TFT is waiting for an OK and do not receive it, it waits for some time (2 min I think but perhaps I increased this time). So perhaps wait more that 2 min before resetting the TFT.

cotepat, 2021-06-03:
No. As soon as I revert to another version it works as it should. I also tested with bluetooth with the exact same results...

mstrens, 2021-06-05:
I just put some changes on github (version of 6 june).
On my test system, I can get Idle status with telnet.
Still, it is not always so each time. Then I select another button (e.g. serial) and then Telnet again.
I do not know why it does not work from the first time.

cotepat, 2021-06-05:
Cool. I will try that!

cotepat, 2021-06-09:
I finally had some time to try it... Here are my findings:

1. As you said telnet now works when switching to wired and back again to telnet. It is not working from start
2. Arrows (move buttons) now work on telnet as they should
3. USB is still disabled as soon as the ethernet cable is connected

So unfortunately it is still not working.

I have an idea (which would be a workaround, temporary solution)... In the connections menu on the controller, could we put a "Disconnect" button? It would do a software disconnect of the TFT and then I could use the computer through GRBL USB port. Then, I could reactivate the connection to the TFT when I want to use it. At least, I would not have to physically disconnect the TFT... I think that would work because, it seems that the only time I had the USB working at the same time as the TFT is with the "??" status.

mstrens, 2021-06-10:
I already tried to disconnect the Serial when Telnet/BT is activated.
The issue is that there is a bug in  Serial.end() arduino function (at least in release 1.0.4).
There is a fix for this issue and I tried to apply it in my code.
I tested it and it seems to work.
I put the version on Github.

cotepat, 2021-06-14:
Yes it does work!! The only problem is you still have to switch to USB(serial) and back to telnet. It does not work from the start. That is not a big deal... thank you so much for your work.

closed 2023-01-16 by cotepat