mstrens/grbl_controller_esp32 #50: compiling failed: SdFat.h: No such file or directory

QGB, 2022-04-24:
```
grbl_controller_esp32:80:10: fatal error: SdFat.h: No such file or directory
 #include "SdFat.h"
          ^~~~~~~~~
compilation terminated.
exit status 1
SdFat.h: No such file or directory
```

QGB, 2022-04-24:
package manage :  install sdfat 

problem solved!

mstrens, 2022-04-28:
Perhaps the issue is related to the version that you are using.
In arduino ide, please use version 1.0.4 of ESP32 extension
For sdfat, use version 2.0.6
For more info, please look at this link : https://www.makerfr.com/cnc/grbl-32bits-board-v2/firmwares-grbl-32bits-board-v2/

QGB, 2022-05-02:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: 86b6856493d23fe741a1623c3d149a38bdf91ca0101173d85f25c3fdb6d605ee
Expected: e7bbd322b12b609ed444a0547cebdc54bbcf46aefed1e04449d9827791f3e8ab
Attempting to boot anyway...
entry 0x403ce000
E (233) gpio: gpio_set_level(226): GPIO output gpio_num error
E (234) gpio: gpio_set_level(226): GPIO output gpio_num error
E (234) gpio: gpio_set_level(226): GPIO output gpio_num error
E (240) gpio: gpio_set_level(226): GPIO output gpio_num error
E (359) gpio: gpio_set_level(226): GPIO output gpio_num error
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380000
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: 86b6856493d23fe741a1623c3d149a38bdf91ca0101173d85f25c3fdb6d605ee
Expected: e7bbd322b12b609ed444a0547cebdc54bbcf46aefed1e04449d9827791f3e8ab
Attempting to boot anyway...
entry 0x403ce000
E (238) gpio: gpio_set_level(226): GPIO output gpio_num error
E (239) gpio: gpio_set_level(226): GPIO output gpio_num error
E (239) gpio: gpio_set_level(226): GPIO output gpio_num error
E (245) gpio: gpio_set_level(226): GPIO output gpio_num error
E (364) gpio: gpio_set_level(226): GPIO output gpio_num error
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: 86b6856493d23fe741a1623c3d149a38bdf91ca0101173d85f25c3fdb6d605ee
Expected: e7bbd322b12b609ed444a0547cebdc54bbcf46aefed1e04449d9827791f3e8ab
Attempting to boot anyway...
entry 0x403ce000
E (238) gpio: gpio_set_level(226): GPIO output gpio_num error
E (239) gpio: gpio_set_level(226): GPIO output gpio_num error
E (239) gpio: gpio_set_level(226): GPIO output gpio_num error
E (245) gpio: gpio_set_level(226): GPIO output gpio_num error
E (364) gpio: gpio_set_level(226): GPIO output gpio_num error
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: 86b6856493d23fe741a1623c3d149a38bdf91ca0101173d85f25c3fdb6d605ee
Expected: e7bbd322b12b609ed444a0547cebdc54bbcf46aefed1e04449d9827791f3e8ab
Attempting to boot anyway...
entry 0x403ce000
E (238) gpio: gpio_set_level(226): GPIO output gpio_num error
E (239) gpio: gpio_set_level(226): GPIO output gpio_num error
E (239) gpio: gpio_set_level(226): GPIO output gpio_num error
E (245) gpio: gpio_set_level(226): GPIO output gpio_num error
E (364) gpio: gpio_set_level(226): GPIO output gpio_num error
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380000
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: 86b6856493d23fe741a1623c3d149a38bdf91ca0101173d85f25c3fdb6d605ee
Expected: e7bbd322b12b609ed444a0547cebdc54bbcf46aefed1e04449d9827791f3e8ab
Attempting to boot anyway...
entry 0x403ce000
E (238) gpio: gpio_set_level(226): GPIO output gpio_num error
E (239) gpio: gpio_set_level(226): GPIO output gpio_num error
E (239) gpio: gpio_set_level(226): GPIO output gpio_num error
E (245) gpio: gpio_set_level(226): GPIO output gpio_num error
E (364) gpio: gpio_set_level(226): GPIO output gpio_num error
```