mstrens/grbl_controller_esp32 #53: text section exceeds available space in board

Henrikastro, 2022-07-29:
Trying to compile hits me with the following message:

> text section exceeds available space in boardO sketch usa 1543542 bytes (117%) de espaço de armazenamento para programas. O máximo são 1310720 bytes.
> 
> Variáveis globais usam 86476 bytes (26%) de memória dinâmica, deixando 241204 bytes para variáveis locais. O máximo são 327680 bytes.
> Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
> Erro compilando para a placa DOIT ESP32 DEVKIT V1

Am I doing something wrong?

Henrikastro, 2022-07-29:
Changed SDFat to 1.1.0 and that fixed it

Henrikastro, 2022-08-03:
Actually, I was able to run using SDFat 1.1.0 on a release version, on the master one, when using SDFat 2.0.6 as mentioned somewhere around here, I get hit with this error. Is it my ESP32 that doesnt have enough memory or am I doing something wrong? I have a DOIT DEVKIT ESP32 and I'm using the 1.0.4 version of ESP32 on Arduino.

regevt, 2023-03-07:
> Actually, I was able to run using SDFat 1.1.0 on a release version, on the master one, when using SDFat 2.0.6 as mentioned somewhere around here, I get hit with this error. Is it my ESP32 that [...]

did you manage to solve the issue? I also having it with the same board