mstrens/grbl_controller_esp32 #63: src/com.cpp:586:13: error: 'gcvt' was not declared in this scope

cbcy1234, 2023-05-24:
src/com.cpp:586:13: error: 'gcvt' was not declared in this scope

darjerz, 2024-04-10:
/mnt/roboczy/cnc/offline/grbl_controller_esp32-master/grbl_controller_esp32/com.cpp:564:13: error: 'gcvt' was not declared in this scope
             gcvt(savedWposXYZA[2], 3, floatToString); // convert float to string
   exit status 1

Compilation error: 'gcvt' was not declared in this scope

kangz543g, 2024-06-07:
delete gcvt(savedWposXYZA[2], 3, floatToString); // convert float to string

#include <string> // Include necessary header for std::string
#include <sstream> // Include necessary header for std::stringstream
#include <iomanip> // Include necessary header for std::setprecision

void sendFromString() {
    char strChar;
    float savedWposXYZA[4];
    waitOkWhenSdMillis = millis() + WAIT_OK_SD_TIMEOUT;

    while (*pPrintString != 0 && (!waitOk) && statusPrinting == PRINTING_STRING) {
        strChar = *pPrintString++;
        if (strChar == '%') {
            strChar = *pPrintString++;
            switch (strChar) {
                case 'z':
                    preferences.putFloat("wposZ", wposXYZA[2]);
                    break;
                case 'X':
                    toGrbl(G30SavedX);
                    break;
                case 'Y':
                    toGrbl(G30SavedY);
                    break;
                case 'Z':
                    savedWposXYZA[2] = preferences.getFloat("wposZ", 0);
                    {
                        std::ostringstream ss;
                        ss << std::fixed << std::setprecision(3) << savedWposXYZA[2];
                        std::string floatToString = ss.str();
                        const char* charPtr = floatToString.c_str();
                        toGrbl(charPtr);
                    }
                    break;
                case 'M':
                    toGrbl(modalAbsRel);
                    toGrbl(modalMmInch);
                    break;
            }
        } else {
            if (strChar != 13 && strChar != ' ') {
                toGrbl((char)strChar);
            }
            if (strChar == '\n') {
                waitOk = true;
            }
        }
    }

    if (*pPrintString == 0) {
        statusPrinting = PRINTING_STOPPED;
        fillStringExecuteMsg(lastStringCmd);
        updateFullPage = true;
        toGrbl((char)0x0A);
    }
}

The sketch uses 1884961 bytes (59%) of program storage space. Up to 3145728 bytes.
Global variables use 89908 bytes (27%) of dynamic memory, leaving 237772 bytes of local variables. The maximum is 327680 bytes.

bdureau, 2026-03-05:
This is due of the version of the ESP32 which has to be 1.04 , older version break