mstrens/grbl_controller_esp32 #67: Compiled successfully but cannot be used, cannot communicate with Arduino + CNC Shield + GRBL 1.1h

satriamahen, 2024-07-13:
All the features from ESP32 work as intended (Display, Wifi, Touchscreen)
But it cannot work with **Arduino GRBL 1.1h + CNC Shield V3.0**
The status on the TFT shows only "??"
All the commands are not working (JOG (Move), Home, Unlock, etc)
I'm connecting the **RX2** from _ESP32 (gpio 16)_ to the **TX** from the _CNC Shield_, and the **TX2** from _ESP32 (gpio 17)_ to the **RX** from the _CNC Shield_.
Did anyone get the same problem?
Am I missing something?

malhar-24, 2025-01-01:
how to compile the project,there are so many error ,i tried platfromio, espidf, ardiuno ide

MrBrain-YT, 2025-11-07:
Same problem, how to solve it?

malhar-24, 2025-11-07:
I did solve and compile this and run it in Arduino ide, there are version of lib that we school  use but i  don't remembers it now,  but has i want the custom gcode generator option i have use below code with esp33 without tft display but an web app to control the grbl Arduino even work with grbl esp32 fuildnc, and i will sujjest you to use web server which is easy to edit and add more feature which you can't do in this contoller code
https://github.com/malhar-24/ESP32-GCode-Transmitter.git